portapack-mayhem/mayhem-firmware #2568: Add an image viewing app

Redgrenn, 2025-03-18:
### Description of the feature you're suggesting.

A new app that could be utilized to view jpg, png, or pdf files. Preferable functions of the app would include:

- zooming in and out
- tracking/panning left, right, up and down
- capable of handling large sized files, such as 75 or 100mb+
- a "center" cursor or marker of some sort

### Anything else?

My personal desire to have this stems from wanting offline high-resolution maps that I could navigate with - separate of anything GPS driven. This could also be used for viewing visual "notes" that a user might want to keep on their device

htotoo, 2025-03-22:
If you save it as a BMP, it is possible now. (not the map part).
Additional file formats sadly no. Much more usable on a phone.

Redgrenn, 2025-03-22:
> If you save it as a BMP, it is possible now. (not the map part). Additional file formats sadly no. Much more usable on a phone.

how? If you mean by saving BMP images for splash screens, if you use a file that is larger than the specified requirements it says "not a valid splash image"

Redgrenn, 2025-03-22:
> If you save it as a BMP, it is possible now. (not the map part). Additional file formats sadly no. Much more usable on a phone.

if you mean that the user could save it and then view it as a BMP under the splash folder, that only works within the parameters available. When attempting to view a large BMP image there, it just says "not a valid splash image"

htotoo, 2025-03-22:
Don't save the bmp to the splash folder. Save to anywhere else.

Redgrenn, 2025-03-22:
> Don't save the bmp to the splash folder. Save to anywhere else.

 I got this method to "work." it loads bmp files, but takes forever and is quite cumbersome and time consuming. If it'd be possible to make an app that could look at images with a similar accessibility and ease of navigating as the worldmap, that'd be great. Not sure how much is wanted/needed to start a bounty, but i'd put $20USD towards this

htotoo, 2025-03-22:
The sad part is BMP is the least computing power neady format.
And that is so slow. Imagine that with a much more complex format, like jpg. Would take forever to load and show, ...

Tldr: doable: maybe, yes.
Worth it? No, since would'n be usable.

zxkmm, 2025-03-23:
As addition of what HTotoo told:

If what all you need is just view world map, then you can just use ADSB Tx app to view without turning on radio.

The reason world map load "faster" than bmp is because world map is a dedicate file with bin extension, it is not an image.

closed 2025-03-23 by zxkmm