portapack-mayhem/mayhem-firmware #2569: Headset Microphone DOA twice.

SpencahDBest, 2025-03-18:
### Describe the bug.

With the H2+ (I know, I read the wiki):
Mayhem Version 2.1.0, flashed with HackRF.app, no major issues with the flash/update (minor requirements to repeat, reinstall some apps). On Zorin 17.2
Used three different known-good headsets.
Same issue on 2 H2+ - returned the first one as defective, same issue in the replacement.
WM8731 Codec (if that matters).

The Issue: Headset plugged in, VU meter does not register voice, TX of the roger beep works as verified on CB radio (really weak signal, but it's there). No voice TX.
VU meter refuses to show any microphone input, TX offers no microphone input.

It would appear that the microphone does not work.

I thought there might be a setting in tx_mic.ini - something to tell it that there is a microphone on that headset? 

What is "mic_mod_index"? It is set to 2, could that be it?
Is there a list of what each of the parameters in the setting file mean - didn't find one on the wiki.

How to make the headset microphone work?

Thank you.

### Reproduction

1. Plug headset into H2+
2. Open Microphone App
3. See nothing on the VU meter, hear no voice through TX.

### Expected behavior

1. See voice register on the VU meter
2. Hear voice TX on different radio at the same freq.

SpencahDBest, 2025-03-18:
Forgot to add that enabling "Hear Mic" also has no effect.

zxkmm, 2025-03-19:
Please make sure your headphone plug type match the portapack socket type. We've seen reports that some H2 has CITA plug, while most H2 has OMTP

[image]

[image]

SpencahDBest, 2025-03-19:
Thank you for that information, if my Headset is CITA (USA standard), would the audio out - speakers - still function on the OMTP?

I still get audio sound out - the headset speakers work, only the microphone doesn't. The ground being common between the mic and the speakers makes me think that both would fail if there is a CITA plug in an OMTP jack - not just one.

SpencahDBest, 2025-03-19:
I ordered an adapter - will update if it works.

zxkmm, 2025-03-19:
If problem is from CITA and OMTP mismatch, the audio would still speak out in weird way: it wires stereo channels together in reverse phase, so if you were playing a song, you can only hear a part of the audio, usually only human voice or only accompaniment.

zxkmm, 2025-03-19:
Sorry I have one thing to correct:
> We've seen reports that some H2 has CITA plug, while most H2 has OMTP

I type them reversed. it should be
> We've seen reports that some H2 has OMTP plug, while most H2 has CITA

SpencahDBest, 2025-03-19:
Thank you for the response. I was able to determine that detail from Google - but seeing as the only thing I've listened for in the H2+ has been static, I wouldn't be able to determine from audio alone. The adapter was only a few dollars, and will come tomorrow - plus it adapts both ways, so it would work in either configuration.

I hope this is the issue - never occurred to me that there might be an issue with the headset pinout, so learn something new I guess.

I will update soon - but I would suggest (maybe I should do it) adding a flag in the Wiki regarding this issue with the H2+ from Amazon.

SpencahDBest, 2025-03-21:
I purchased an adapter, no good. It does blink one bar on the vu meter when I mute the mic, but still no signal in.

Could this be related to an ALC issue from 1.4.0 from 2022? I saw that, but didn't really understand it. User reported no audio after using the microphone app, then no vu meter activity, so I thought maybe...

Any ideas or is this thing a 300$ brick? Not much good for a transmitter with no mic 😔

zxkmm, 2025-03-21:
I read the issue #620 you mentioned, and it's been fixed already, so i think it might not from there.

> but seeing as the only thing I've listened for in the H2+ has been static

Could you pls explain this with details? You meant you can't even hear local radio?
If so then maybe broken audio chip?

Also, would you mind sharing any details of the issue, for example from which firmware version it stopped working? 
Since you mentioned 300USD H2+, I assume you already have it for a while? (Because h4m is like ~200 USD now), So did it ever work?

SpencahDBest, 2025-03-21:
Your reply was very kind. Thank you.

I just recently bought it from Amazon (https://www.amazon.com/dp/B0DKBLDLK5) - then had it replaced, it's now in its return box for a refund without replacement. After taxes, it was $290.00.

I was able to get local FM radio, not the best reception, but I was in the basement so that's not surprising.
The mic never worked out of the box, on both units (the first one, and the replacement). Tried with three known headsets, tried with a CTIA - OMTP adapter, never got any mic function. Nothing in the VU meter, nothing transmit.
It didn't work with 2.0.1 (arrived with that distro), and it didn't work with 2.1.0. I also tried a random nightly distro, and it didn't work then either.

I'm sad - Mayhem looked really cool, and I was very excited to play with it - but this is just too much aggravation for what it cost. Maybe I might try an H4M, but I'm pretty bummed right now, so I think I'm going to close my Github, and just forget about HackRF for a while.

Thanks again.

zxkmm, 2025-03-21:
You are welcome! 
In this case I think you can just simply return the device, it might be broken in factory already.
I didn't expect that you would delete your account tho, I will re-open this issue as "complete" instead of not-plainned.

closed 2025-03-21 by zxkmm