portapack-mayhem/mayhem-firmware #2571: naming space issue for 4 games

zxkmm, 2025-03-19:
### Describe the bug.

we have currently 4 (tetris and other 3 that created with tetris as template) games that has some code is outside of it's own naming space, leading to they're occupying some inner space.

We should discuss how to make it work.

### Reproduction

1

### Expected behavior

1

htotoo, 2025-03-19:
My idea (not nice, but easy to implement) is to move c functions to a  c++ class (with static functions and variables) that is in the right namespace.

zxkmm, 2025-03-19:
Thanks, I will try them. 
If I may ask, do you think the issue is because of naked C func? 
What would happen if have naked func in correct naming space?

zxkmm, 2025-03-19:
last time i tried to just simply add naming space and it seems won't even compile, im not sure how external app works.

gullradriel, 2025-03-19:
Because you did use the namespace two times.
Look, the whole game is already in a external namespace, the cpp missing a namespace is not directly compiled but included with a #include

gullradriel, 2025-03-19:
Plus I didn't see firmware space being reduced with the new games.
One would have to check firmware size without the games to be sure.

zxkmm, 2025-03-19:
> One would have to check firmware size without the games to be sure.

already checked last time when the snake game came out

zxkmm, 2025-03-19:
with all 4 apps:

[image]

without 4 apps:

[image]

gullradriel, 2025-03-19:
You should try one by one, because all four at a time and only 2k, maybe not all are bad ?

zxkmm, 2025-03-19:
It would be a nice idea to try one by one. 
But based off the fact that the other 3 games were just using Tetris as template, and all of them has same _middle layer files*_ which is out of naming spaces:

- Arial12x12.h
- mbed.h
- SPI_TFT_ILI9341.h

---
*middle layer files: the Tetris game was designed for a game console, and someone ported it into Mayhem, and written some middle layer files to behaving like the virual HAL (make hardware compatible with PortaPack.)

zxkmm, 2025-03-19:
And these 3 middle layer files aren't that big (ablut 100-200 lines totally), that's why they didn't occupy much

htotoo, 2025-03-19:
2k is a lot in this project. 2k can be a lot of new subghzd protocols, ... If we can, we should free that up.

NotherNgineer, 2025-03-19:
> 2k is a lot in this project. 2k can be a lot of new subghzd protocols, ... If we can, we should free that up.

Regarding space savings, I think we should also consider making "Debug Dump" (Pmem Dump) an external app; this would save several KB.

zxkmm, 2025-03-20:
#2577

closed 2025-03-20 by zxkmm