portapack-mayhem/mayhem-firmware #2580: Instability past v1.7.3 OpenSDRLabs H2+ (QFP100)

lvent, 2025-03-21:
### What happened?

I am getting red screens, hard locks, stuck on various screens, stuck on initialization of different components (like sd card). None of these happen on 1.7.3 however it starts happening with 1.8.0. I found that most reliable way to test if its going to happen is that scrolling in Looking Glass presets, device locks up or not.

This device arrived with 1.7.3 from Aliexpress OpenSDRLabs. Looking inside, it has an QFP100 package CFLD on it.

- I tried flashing Jumbo77 1.43, then flashing everything back via DFU and HackRF mode.
- Different SD cards, different partitioning sizes, different clustering sizes.
- I tried the nightly.
- Charging the battery overnight
- Checking battery voltage,
- removing battery and using with a 3amp charger (tested with an external load that it does 3amps)
- Reseting persistent memory
- Disabling fast sd card transfers.
- Removing clock battery

Flashing Jumbo77s 143 turns the device into a brick, I only get rx light flashing, however afterwards I can flash official mayhem firmwares and portapack boots back again albeit with these problems.

Any ideas? I have been pulling my hair over this device for a while now and I cannot think something else.

### Your hardware

OpenSDRLabs H2+ (QFP100)

### Tests

- [x] I managed to update/boot previously

gullradriel, 2025-03-21:
> I tried flashing Jumbo77 1.43, then flashing everything back via DFU and HackRF mode.

That's an error. Jumbo77's firmware is a closed source firmware, which only works on the device he sold. Maybe it can work on the other devices, but we are not giving any support to that firmware.

> Disabling fast sd card transfers

Enabling it is only working on some H1 AFAIK

What I would try;
- use the nightly
- use up to date SDCARD contents
- update CPLD (see https://github.com/portapack-mayhem/mayhem-firmware/wiki/Updating-the-Xilinx-CPLD-on-hackrf-board)

gullradriel, 2025-03-21:
Note: I also have a H2 from OSSDRLabs and do not have the problems you're having.

lvent, 2025-03-21:
I forgot to mention that I also tried that. Just for good measure did it once again, I will update.

gullradriel, 2025-03-21:
Next step is to ask OpenSourceSDLabs what can be the difference. 
As you can see you're the first to open such an issue, so it may be your device.

lvent, 2025-03-21:
Nope same thing, happens. 1 out 3 boots results in device lock up. 1 out 3 times I attempt to use looking glass, device locks up, when it launches, switching between presets in looking glass results in hard locks.

gullradriel, 2025-03-21:
Go into utilities, Debug, and:
- use Debug Dump, upload the file
- screen the contents of 'Peripherals' and a screen of each sub menu
Also please dismantle the device, take photos of both side of each board.
Maybe we can help, maybe not.
Posting those informations may give some hints.

lvent, 2025-03-21:
[DEBUG_DUMP_0000.TXT](https://github.com/user-attachments/files/19394720/DEBUG_DUMP_0000.TXT)

[image]
[image]
[image]

lvent, 2025-03-21:
[image]
[image]
[image]
[image]
[image]

gullradriel, 2025-03-21:
RFFC5072 looks identical
[image]
**⚠ But my MAX2839 is not having the same values ⚠** 
[image]
The Si5351A looks identical
[image]
The WM8731 looks identical 
[image]

gullradriel, 2025-03-21:
It may be normal. I'm on nightly 250121 on that device. But it's working fine.
A dying coin cell can be the problem. You should also check it / check that the coin cell holder is not badly soldered.

lvent, 2025-03-21:
Put in a brand new cell just now, same thing sadly. I also tried updating those values to be the same as yours and it seems I cant. Problem sadly still persists and goes away when I downgrade to 1.7.3. It even happens with 1.8.0 as well.

gullradriel, 2025-03-21:
Well. Usually these kind of problems were related to the CPLD.
But you told you updated it.
Maybe try, if not already:
- disassemble
- update hackrf and hackrf CPLD without portapack on it
- reassemble
- be sure to only have the sdcard content on the sdcard and that the sdcard isn't faulty
- update to latest nightly
- do a pmem reset
- reboot

gullradriel, 2025-03-21:
Else you will have to track at which point exactly it's stopping.
1.7.3 is a stable, you would have to try each nightly after the 1.7.3 and tell us when you have the exact nightly version that is making the problem appear. 
Only then we can track down the bug using the commits/ diffs.

gullradriel, 2025-03-21:
1.7.3 is on page https://github.com/portapack-mayhem/mayhem-firmware/releases?page=36
Next nightly is [Nightly Release - 2023-07-02](https://github.com/portapack-mayhem/mayhem-firmware/releases/tag/nightly-tag-2023-07-02)
Please go up and up in the release page, install each nightly and test.
If you can track down at which version exactly it's crashing, maybe we can help more.

gullradriel, 2025-03-21:
Using that filter you can try with just the stables https://github.com/portapack-mayhem/mayhem-firmware/releases?q=%22Stable+release%22&expanded=true
And then narrow down to the nightly causing the problem between the last working stable and the faulty one.

Brumi-2021, 2025-03-22:
Hello are you sure that your device was completely working well with our original 1.7.3 ? 
If you are not sure , change back to that original fw condition to test it . I think that  If your set worked perfect with mayhem  1.7 3 , it should also work with current mayhem fw .

lvent, 2025-03-22:
I tested around approximately 40 nightlies between 1.7.3 and latest 2.1.0 last night. Between 1.7.3 and 1.7.4 I noticed that I would start to get occasional crashes with 230909 but still no looking glass crashes.

Then I got tired of it and flashed the 2.1.0 with complete SD card package. I noticed that some tasks are super slow (like when viewing the map in ADSB page) and upon booting sometimes it takes good 15 seconds to get to mayhem firmware splash screen.

I cannot put my hand on what is causing what, but at this point I am certain it has to do with FULL SD package as well. I basically wasted 3 hours of my life with minimal packages last night.

I will close this issue for now, I will update if there is anyhting new comes up.

Brumi-2021, 2025-03-22:
My point is that , 
you need ro be sure if all was stable with original  shipping conditions fw 1.7.3 . (And re confirm it again ) 

If not , then we will doubt about hardware , and my advice ia ask to change the faulty unit to the seller .

But If all was ok with this original condition, then we should find the way to make it work with latest mayhem fw . 

FYI 
Long time ago i had many accidental bricks in my other PP qfp64 h1 version (with rottary buttons) related to my different version gcc arm compiler and the pmem , and i need to frequenly come back to the very old  1.4.3 mayhem fw (without pmem)  to recover the unit . But all those headaches were gone , thanks to @bernd-herzog who make a better boot init process able to check automatically and repair my white flash bricked unit .  So nowadays i do not have any brick issue with my units

zxkmm, 2025-03-23:
I think it’s not needed to close if the issue isn’t resolved.

zxkmm, 2025-03-23:
> sometimes it takes good 15 seconds to get to mayhem firmware splash screen

Which icon it stuck at, during the booting sequence?

lvent, 2025-03-23:
Changes each time, sometimes during radio, sometimes during sd card. I have been using the device extensively since yesterday and other than maps in ADSB reader being extremely slow and occasionally taking 10 to 15 seconds to respond when viewing ADSB map, I have not experienced anything else.

Same SD card with 1.7.3 has no problems whatsoever.

JamesVStone, 2025-04-21:
I am having the exact same issue on a portapak H2 from OSSDRLabs

zxkmm, 2025-04-21:
Please confirm this:
If you back to 1.7.3, both firmware and FULL sdcard content, would it become normal again, or it’s still glitch anyhow?

Just like Brumi said, this is very important for us to debug, but OP didn’t provide any info of this.

Also, plz tell have you used jumbo’s fw too? Because we don’t know what he did in the firmware, he could’ve made some permanent changes that out of our control.

JamesVStone, 2025-04-21:
Just loaded 1.7.3 via flash utility with full SD image and it works flawlessly.

1.7.4 is what is was using prior to 2.1.0 and that also worked fine.

I have not used Jumbo's firmware to my knowledge. I originally used a version of mayhem_skip firmware which was provided by OSSDRLab in a fileshare.

zxkmm, 2025-04-23:
Anyway @JamesVStone if you have time, I think it would be nice if you can try each nightly and found out which one made it abnormal. 

Because we can't reproduce the issue on our own devices, and also we don't have controls of what your device had been through, we are not able to debug this issue.

Another option is you ship your device to developers who's volunteers helping to debug the issue. But I doubt if anyone willing to and also if anyone lives near you.

lvent, 2025-04-23:
I should probably chime in here. I spent HOURS flashing and testing nightlies, you cannot replicate this issue with just minimal packages. It has something to do with full package (with the SD files). My device somehow currently works with the full SD image but somethings related to SD card are extremely slow.

ADSB map data loads in 30 seconds or more, when in there, device gets super laggy and response time is like 10 seconds or more. I tried different SD cards and the same deal.

zxkmm, 2025-04-23:
Sorry but I don’t understand what does “ you cannot replicate this issue with just minimal packages” means.

Are you saying that you can’t replicate the issue in “minimal” change (nightly) or are you saying that you can’t always replicate the issue (sometimes it behaves normal and sometimes not)?

zxkmm, 2025-04-23:
The only solution (not a real solution but more like a start that we can fetch valid information) would be:

Find out which commit (AKA code change/edit/add/delete) caused that issue, so we can dig more.

gullradriel, 2025-04-23:
> I should probably chime in here. I spent HOURS flashing and testing nightlies, you cannot replicate this issue with just minimal packages. It has something to do with full package (with the SD [...]

Check in Settings / Sdcard that high speed mode is NOT activated. It's known to cause problems on devices not supporting it.

gullradriel, 2025-04-23:
Install nightyl, clean out SETTINGS/ , got to Settings/Pmem menu, clear persistent memory. 
Check that CPLD is up to date, and that sdcard high speed mode is not activated.

JamesVStone, 2025-04-24:
I loaded the full sdcard image of nightly (no world map as all other sdcard images I've downloaded have been) `250423` and flashed the firmware from utilities -> flash and it was working consistently with no crashes.

I then cleared the PMEM under gullradriel's suggestion and reloaded the sdcard with v2.1.0 but then it crashed - **without** flashing the firmware from v2.1.0!

So the v2.1.0 sdcard files caused the nighly firmware to crash.

I believe that this is an issue with the sdcard files included with v2.1.0 not the PMEM.

I will have time this weekend to test nighlys one by one to track down this bug if helpful.

zxkmm, 2025-04-25:
If you already found the reason of crash is sdcard content, then you don’t need to flash nightly one by one anymore.