portapack-mayhem/mayhem-firmware #2592: HackRF_sweep not able to kill the process it self

MaRcOsss1, 2025-03-25:
### Describe the bug.

When running hackrf_sweep on my HackRF One, I am experiencing intermittent failures where no data is transferred. Sometimes the command completes successfully, but other times it exits with the error: Couldn't transfer any data for one second.
**System Information:**

- Hardware: HackRF One
- Software Version: Latest hackrf-tools (installed from official repo)
- OS: Ubuntu 22.04
- Command Used: `hackrf_sweep -f 2400:2500 -N 2` (or anynhackrf_sweep command with changed parameters)
- Sample Rate: 20 MHz
- Baseband Filter Bandwidth: 15 MHz

Log:
```
aryan@SDR:~$ hackrf_sweep -f 2400:2500 -N 2
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 2400 MHz to 2500 MHz
Stop with Ctrl-C
0 total sweeps completed, 0.00 sweeps/second

Couldn't transfer any data for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total sweeps: 0 in 1.01863 seconds (0.00 sweeps/second)
hackrf_close() done
hackrf_exit() done
exit
aryan@SDR:~$ hackrf_sweep -f 2400:2500 -N 2
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 2400 MHz to 2500 MHz
Stop with Ctrl-C
0 total sweeps completed, 0.00 sweeps/second

Couldn't transfer any data for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total sweeps: 0 in 1.01546 seconds (0.00 sweeps/second)
hackrf_close() done
hackrf_exit() done
exit
aryan@SDR:~$ hackrf_sweep -f 2400:2500 -N 2
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 2400 MHz to 2500 MHz
Stop with Ctrl-C
2025-03-25, 06:23:53.140556, 2400000000, 2405000000, 1000000.00, 20, -62.37, -50.34, -39.07, -40.67, -63.25
2025-03-25, 06:23:53.140556, 2410000000, 2415000000, 1000000.00, 20, -56.18, -63.36, -64.67, -60.87, -58.81
2025-03-25, 06:23:53.140556, 2405000000, 2410000000, 1000000.00, 20, -65.37, -62.69, -60.48, -56.86, -56.08
2025-03-25, 06:23:53.140556, 2415000000, 2420000000, 1000000.00, 20, -56.26, -58.17, -63.26, -67.96, -63.00
2025-03-25, 06:23:53.140556, 2420000000, 2425000000, 1000000.00, 20, -62.16, -59.11, -58.39, -59.89, -71.63
2025-03-25, 06:23:53.140556, 2430000000, 2435000000, 1000000.00, 20, -57.44, -66.20, -63.35, -86.36, -60.34
2025-03-25, 06:23:53.140556, 2425000000, 2430000000, 1000000.00, 20, -49.38, -54.13, -64.31, -63.02, -65.65
2025-03-25, 06:23:53.140556, 2435000000, 2440000000, 1000000.00, 20, -56.07, -53.67, -56.63, -54.70, -54.85
2025-03-25, 06:23:53.140556, 2440000000, 2445000000, 1000000.00, 20, -61.39, -60.72, -65.95, -61.19, -63.21
2025-03-25, 06:23:53.140556, 2450000000, 2455000000, 1000000.00, 20, -60.22, -57.37, -61.87, -64.06, -66.02
2025-03-25, 06:23:53.140556, 2445000000, 2450000000, 1000000.00, 20, -61.08, -63.25, -64.68, -58.74, -55.93
2025-03-25, 06:23:53.140556, 2455000000, 2460000000, 1000000.00, 20, -53.56, -53.65, -58.53, -60.66, -67.74
2025-03-25, 06:23:53.140556, 2460000000, 2465000000, 1000000.00, 20, -57.91, -59.32, -63.89, -62.11, -60.57
2025-03-25, 06:23:53.140556, 2470000000, 2475000000, 1000000.00, 20, -59.65, -56.10, -56.57, -64.42, -59.75
2025-03-25, 06:23:53.140556, 2465000000, 2470000000, 1000000.00, 20, -59.04, -57.31, -56.17, -57.26, -56.81
2025-03-25, 06:23:53.140556, 2475000000, 2480000000, 1000000.00, 20, -59.02, -70.80, -64.85, -77.61, -55.46
2025-03-25, 06:23:53.140556, 2480000000, 2485000000, 1000000.00, 20, -57.05, -58.60, -54.93, -56.45, -64.66
2025-03-25, 06:23:53.140556, 2490000000, 2495000000, 1000000.00, 20, -73.19, -65.34, -58.65, -57.51, -56.79
2025-03-25, 06:23:53.140556, 2485000000, 2490000000, 1000000.00, 20, -61.99, -63.18, -61.64, -56.32, -63.19
2025-03-25, 06:23:53.140556, 2495000000, 2500000000, 1000000.00, 20, -60.49, -59.72, -59.46, -62.93, -61.57
2025-03-25, 06:23:53.140556, 2400000000, 2405000000, 1000000.00, 20, -53.95, -55.93, -60.60, -60.21, -61.59
2025-03-25, 06:23:53.140556, 2410000000, 2415000000, 1000000.00, 20, -57.72, -61.06, -64.87, -57.42, -53.98
2025-03-25, 06:23:53.140556, 2405000000, 2410000000, 1000000.00, 20, -54.47, -61.65, -56.68, -58.97, -59.51
[... 177 lines cut ...]
Total sweeps: 0 in 1.01589 seconds (0.00 sweeps/second)
hackrf_close() done
hackrf_exit() done
exit

```

### Reproduction

1. Power on HackRF One and connect it to the system.
2. Open terminal on the connected system.
3. Run command: `hackrf_sweep -f 2400:2500 -N 2` for 5-7 times 
4. Observe output for failure message: `Couldn't transfer any data for one second.`
5. Repeat command multiple times (3-5 times) to check for inconsistencies.
6. Check for consecutive failures (multiple runs with 0 total sweeps completed).

What I did:
- Ran the command: Spectrum sweep completed, data received, and sweep stopped.
- Ran the command again: Encountered a failure.
- Ran the command again: Worked successfully, spectrum sweep completed, data received, and sweep stopped.
- Ran the command a fourth time: Encountered failure again.

### Expected behavior

I expected the hackrf_sweep -f 2400:2500 -N 2 command to reliably perform a spectrum sweep, stop, and then immediately work again each time I rerun the command without intermittent failures.

gullradriel, 2025-03-25:
You have opened the issue in the wrong GitHub.
Here it's the portapack mayhem GitHub.
For all issues in hackrf mode or just using a hackrf, please open the issues on hackrf GitHub:
https://github.com/greatscottgadgets/hackrf

closed 2025-03-25 by gullradriel

zxkmm, 2025-03-25:
Please open a issue ticket in hackrf repo. We don't maintain hackrf code.