portapack-mayhem/mayhem-firmware #2603: DFU firmware flashing not working as documented

sgstair, 2025-03-27:
### Describe the bug.

Checked against release v2.1.0 and v2.0.2 (though looks like the dfu/flashing tools haven't changed in quite some time)
Tested on Windows 10, but I'd expect other windows versions to behave in the same way.
Based on the instructions:
> If you are in Windows, from the release package double click dfu_hackrf_one.bat and follow the instructions. Do not disconnect or reset your PortaPack after that procedure, run flash_portapack_mayhem.bat.
(https://github.com/portapack-mayhem/mayhem-firmware/wiki/Update-firmware-troubleshooting)

I had a HackRF that had a bad/corrupt firmware and did not boot correctly.
I was able to boot the HackRF in DFU mode and loaded the RAM-resident HackRF loader image with the first script, but the second one did not detect a HackRF device.
`Devices detected: 0 DFU, 0 HackRF`
`ERROR: HackRF not found.`

The device was present with (winusb) drivers loaded successfully in my system, labeled as "HackRF One", HWID `USB\VID_1D50&PID_6089\RunningFromRAM`

I was eventually able to flash the firmware by using the `hackrf_spiflash` tool from `hackrf/tools`, it did detect the device and flashed successfully. I was surprised that the tool included with the mayhem release was not able to do this given the instructions on the wiki.

### Reproduction

1. (May need to erase/damage firmware on HackRF, not sure. Probably not necessary)
2. Boot HackRF to DFU mode
3. Run `dfu_hackrf_one.bat` from extracted release package - Don't reset HackRF after this step.
4. Run `flash_portapack_mayhem.bat` from extracted release package. Probably this will fail as described above with `ERROR: HackRF not found.`

### Expected behavior

`flash_portapack_mayhem.bat` should detect the device that enumerated after the previous step and flash firmware rather than failing to detect the device.

zxkmm, 2025-03-27:
try install driver, change different cables / different computer/OS or other stuff that in FAQ.
The dfu is not possible to be broken, we cant touch it, and which also means we can't do anything if it's really broken somehow.

zxkmm, 2025-03-27:
And also you need to correctly put into DFU mode to flash with DFU fw. 
Same that you have put into hackrf mode to flash with hackrf_spiflash tool

sgstair, 2025-03-27:
Where does the `utils/hackrf_update.exe` in your release package come from? it does not seem to be a current hackrf tool, maybe it's an older one?
It's not the DFU that's broken, it's that tool in particular. 

As I said, the hackrf `hackrf_spiflash` tool did work, so clearly the DFU part of the proccess did work, but following the documention from this project (linked) to flash via DFU did not work for me, that's what I'm expressing here.

zxkmm, 2025-03-28:
> Where does the utils/hacker_update.exe in your release package come from?

It’s provided by GSG and maybe the source isn’t available (CMIIW)

> it does not seem to be a current hackrf tool, maybe it's an older one?

It could be outdated but I just tried on windows and it works.

> As I said, the hackrf hackrf_spiflash tool did work, so clearly the DFU part of the proccess did work

hackrf_spiflash doesn’t using DFU to flash. hackrf_spiflash doing the same that utils/hacker_update.exe does

> but following the documention from this project (linked) to flash via DFU did not work for me, that's what I'm expressing here.

What error you are experiencing?

sgstair, 2025-03-28:
>What error you are experiencing?

As stated in the bug report:
1) Booted HackRF to DFU mode using button combination - **Succeeded**, device enumerated as DFU
2) Run `dfu_hackrf_one.bat` from unzipped release package - **Succeeded**, device re-enumerated as "HackRF One"
3) Run `flash_portapack_mayhem.bat` from unzipped release package - **Failed**, did not detect any HackRF One device.
  this uses `utils/hackrf_update.exe` in release package.
  Output includes: 
    `Devices detected: 0 DFU, 0 HackRF`
    `ERROR: HackRF not found.`
  Note that this process is exactly what is documented at https://github.com/portapack-mayhem/mayhem-firmware/wiki/Update-firmware-troubleshooting#windows - It did not work for me.

* Separately sourcing `hackrf_spiflash` and using it to flash the firmware from the release package - **Succeeded**, detected the device that `hackrf_update` did not, and proceeded to flash correctly

zxkmm, 2025-03-28:
You are doing them steps correctly following the guide of the wiki. 

But I just tried again on my windows 10 and I can’t seem to reproduce the issue you’re experiencing. 

I think what the best regards this issue we can do is that try to see if GSG provides an updated binary, and update it.

Or someone that good at windows native development can contribute a open source one. It shouldn’t be hard just wrap the spiflash code one more layer.

But overall I think usb stuff is tricky, even more tricky on windows, so it’s already good enough that you already found a way to fix your devices.

zxkmm, 2025-03-28:
Ok I did some research, and I have to correction what I said that the executable were provided by GSG. 

Since I didn't found that from GSG's repo, so I tried to reverse engineer ing it and was linked with a debug symbol and the symbol says it was compiled by jboone. But the sad part is, jboone doesn't provide source code as well.

sgstair, 2025-03-28:
I did manage to figure out the problem.

The HackRF uses a driverless install method whereby the USB devices tells windows about its requirements and windows loads an appropriate winusb driver.
Unfortunately, windows never updates some details about this device even if the device firmware changes and indicates additional capabilities / interface GUIDs down the road - I have been using HackRF devices for a very long time and apparently some things have been added since the last time I've refreshed the driver.

I could reproduce the bug on my system even after flashing the device - but after uninstalling the device in device manager and rerunning the steps, this `hackrf_update` tool does work now. I'm still a little confused why it failed but the official tools continued to work, but they are probably reyling on a slightly different interface.

So if it comes up again, the solution is either to uninstall the "HackRF One" device in device manager, or upgrade it using the provided INF.

Thanks!

zxkmm, 2025-03-28:
Thank you for pointing out this issue. I guess we have to re-write this tool someday, maybe next week

eried, 2025-03-28:
> So if it comes up again, the solution is either to uninstall the "HackRF One" device in device manager, or upgrade it using the provided INF.
> 
> Thanks!

You never used Zadig on that previous "HackRF One" right?

sgstair, 2025-03-28:
> You never used Zadig on that previous "HackRF One" right?

I don't believe so

closed 2025-03-31 by eried