portapack-mayhem/mayhem-firmware #2604: Modulation switching to AM DSB when it shouldn't

Milka7878, 2025-03-31:
### Describe the bug.

In Audio RX the modulation switches to DSB 9k everytime the focus is getting over the Modulation field.

### Reproduction

Set the modulation to AM USB.
Push up, to get to Frequency input.
Push left to get to Modulation again.
Modulation switches back to AM DSB.

### Expected behavior

Modulation should stay the same when focus is pointing to modulation field.

### Environment/versions

n_250328

gullradriel, 2025-03-31:
Hah. I see it now. Let me search.

Brumi-2021, 2025-03-31:
I could replicate from both DSB AM 6K and 9K
But sometimes it works well . It is a minor issue to be fixed. Thanks for finding it .

Milka7878, 2025-04-04:
Can someone please mark this as a bug? It's definitely not a hardware issue.

Brumi-2021, 2025-04-04:
It was a pure fw issue. 
I think I already solved. 
I submitted PR for its approval,  PR (#2609) 
if you are in Discord, I can submit advanced binary for your testing . 

Thanks,

gullradriel, 2025-04-04:
Fixed in https://github.com/portapack-mayhem/mayhem-firmware/pull/2609