portapack-mayhem/mayhem-firmware #2613: TRX menu apps

strijar, 2025-04-09:
### Description of the feature you're suggesting.

In the firmware, there are applications only for RX and applications only for TX. They have corresponding groups in the menu. But there is an application Mic that works both for RX and TX. I plan to implement several more applications of this type. For example, RX/TX of amateur radio protocols (RTTY, BPSK, etc.). It probably makes sense to create a separate group for such applications.

Brumi-2021, 2025-04-09:
Hi @strijar , i think nice proposal .
And welcome back onboard  ! 

Cheers

strijar, 2025-04-10:
Then I'll take on this task. Maybe it makes sense to think about creating some more additional groups. For example, to work with IQ files (record, replay, view)

Guys, you have done an amazing job during this time! Apparently I needed to step away from things a little to look at things with fresh eyes. I have a lot of ideas and I have already started implementing some of them.