portapack-mayhem/mayhem-firmware #2616: A firmware-user manual mismatch.

Unguvauk, 2025-04-10:
### Description of the feature you're suggesting.

H4M Portapack --> Transmit --> Keyfob&OOKBrute.

Both Apps (Keyfob and OOKBrute) are items in the firmware; I can select them and start them. But I do not understand them. There is no mention of them in the Wiki User Manual, and I couldn´t find a description of the two Apps anywhere.
Please, let me know.

eried, 2025-04-10:
Hi!

The app was removed because was not working, then it was brought back @zxkmm (?) to make it work, but I am unsure of the current state. This is a collaborative project and the wiki is fully open for anyone to add their findings so is not that "someone" forgot to write about this 😅 is just that hasn't been written.

 Options for you:
1. read the code of the app, try to help with code / wiki
2. wait until someone collaborates on these

zxkmm, 2025-04-10:
For keyfob: no idea if it works, but it does have some valid algorithms in the code. This I kept it as external app. (Because some user were pissed when we removed it)

For ookBrute: it is a new app made by @htotoo which works great. Read wiki for more details.