portapack-mayhem/mayhem-firmware #2626: HACKRF ONE H4M UHF RFID SCAN?

ZaBest20, 2025-04-13:
### Description of the feature you're suggesting.

Hi guys,
I'm new in this world, i just bought the portaback hackrf one h4m, i have a RFID UHF semi-passive tag, i use it on my car when traveling to europe to pay the toll roads . When i arrive at the toll place, a beep comes from my tag ( which has a battery) and then the light goes green and the toll barrier goes up and i can pass. Is it possible to detect the info from my tag with the hackrf one? I would really apreciate some help ( step by step ) from somebody that has time. Is there an alternative for the hackrf ?
Thank you.
P.S Here you have the image of the tag

https://postimg.cc/gallery/yghgCQp

ZaBest20, 2025-04-13:
[image]
[image]

zxkmm, 2025-04-14:
Sorry, but the issue section of this project is for posting about firmware bug or firmware enhancement, it’s not for discussion or asking general radio questions.

If you have general radio questions, please join our discord to ask and discuss.

zxkmm, 2025-04-14:
https://discord.gg/9xyM3983