portapack-mayhem/mayhem-firmware #2636: Tetris end-of-game bug

thejapmannz, 2025-04-23:
### Describe the bug.

While playing tetris, it is possible to prematurely trigger losing the round, by rotating a piece just after spawning while it is still at the top of the screen. It seems to only occur with the blue 4x1 piece, which if rotated immediately after spawning can appear to be 2 blocks above the viewable area and 2 blocks within view.

### Reproduction

1. Be playing Tetris app
2. Wait for blue 4x1 block to be next piece to spawn
3. Mash middle button (rotate piece)
4. If rotated early enough, an end-of-game message is triggered 

### Expected behavior

End of game only triggered when stacked pieces reach top of screen, not freshly spawned pieces.

### Environment/versions

Portapack nightly 250423

### Anything else?

Discovered by accident, the buttons on my portapack are very bouncy...

RocketGod-git, 2025-04-23:
I'll take a look at this after work. Thanks for reporting.

RocketGod-git, 2025-05-09:
Fixed in 
[https://github.com/portapack-mayhem/mayhem-firmware/pull/2650](https://github.com/portapack-mayhem/mayhem-firmware/pull/2650)

Please test after nightly update incorporates it and if fixed, close the issue. 😎

closed 2025-05-12 by zxkmm