portapack-mayhem/mayhem-firmware #2639: update firmware White screen not boot?

littleyeson, 2025-04-28:
### What happened?

Update firmware。Reboot white screen. I have tried https://github.com/portapack-mayhem/mayhem-firmware/wiki/Won't-boot  hold up key 10sec before power on。and try another  hold down/left/right/ok key . the above method, but the is still white screen. I take off PortaPack，only use Hackrf( Upload DFU fimware）can work for   in SDRsharp. I try to put PoraPack back of the  Hackrf. and update firmware by https://github.com/portapack-mayhem/mayhem-firmware/wiki/Update-firmware#dfu method. It still white screen. How to fix it. Could I upload the firmware of MCU on the PortPack broad to fix it？

### Tests

- [ ] I managed to update/boot previously

zxkmm, 2025-04-28:
Plz take out sdcard and coin battery and try boot again.

littleyeson, 2025-04-29:
> Plz take out sdcard and coin battery and try boot again.

I try to take out sdcard and coin battery. It is still whitle screen.

zxkmm, 2025-04-29:
https://github.com/portapack-mayhem/mayhem-firmware/wiki/Full-reset

plz read this and tell me which step you stuck on

littleyeson, 2025-04-30:
> https://github.com/portapack-mayhem/mayhem-firmware/wiki/Full-reset
> 
> plz read this and tell me which step you stuck on

I have

> https://github.com/portapack-mayhem/mayhem-firmware/wiki/Full-reset
> 
> plz read this and tell me which step you stuck on

I have already taken all the methods in this link before. I think it may be necessary to rewrite the MCU firmware on PortaPack's back . But the specific operation method and firmware content cannot be found. The firmware only has hackrf one

zxkmm, 2025-04-30:
写cpld的教程可以自己找，都是公开的。

如果是厂家想量产的话建议不要伪装成普通用户设备坏了，因为太明显了。而且浪费开发者的时间。

closed 2025-04-30 by zxkmm