portapack-mayhem/mayhem-firmware #2640: Real Time Clock - Scheduling transmitter

Tuxedosam63, 2025-04-30:
### Description of the feature you're suggesting.

Hello,
Having functionality to schedule some operations like transmitting a signal at a specific time/day using the real time clock on the H4M would be great. I had asked on the discord but was told that this functionality does not currently exist.

htotoo, 2025-04-30:
Via serial usb, you can do it, but you'll need a pc / raspberry. the portapack can't do it by itself.

zxkmm, 2025-05-02:
this app sounds interesting, i will make it

eried, 2025-05-06:
> this app sounds interesting, i will make it

for scheduled pranks? 🙄

zxkmm, 2025-05-06:
Ya exactly

zxkmm, 2025-05-06:
It actually indeed made me thinking what this could be used at, in legally scope?

It is a nice idea indeed. But I can't imagine a legal usage for this. OP would you mind give me a hint?

gullradriel, 2025-05-06:
> It actually indeed made me thinking what this could be used at, in legally scope?
> 
> It is a nice idea indeed. But I can't imagine a legal usage for this. OP would you mind give me a hint?

Opening and closing old electric curtains at specific hours ? X-D

Tuxedosam63, 2025-05-06:
> It actually indeed made me thinking what this could be used at, in legally scope?
> 
> It is a nice idea indeed. But I can't imagine a legal usage for this. OP would you mind give me a hint?

Main motivation I am requesting this feature is for automatic control over my automatic bedroom blinds, adjustable bed frame, and portable AC unit all controlled by remotes with no app integration. Being able to control these at specific times and days would be extremely useful for morning routines in my case. I can imagine other ways it would serve people well too.

Tuxedosam63, 2025-05-06:
> > Opening and closing old electric curtains at specific hours ? X-D

Unironically yes lmao

RocketGod-git, 2025-05-06:
Sprinkler timer, Nightly garage close, turn off the kids tv at bedtime, turn off asshole neighbor lady's tv every 5 minutes 🫣🤪🙃

zxkmm, 2025-05-07:
Sure then, I’ll make it

zxkmm, 2025-05-07:
I’m not sure if the long run stability tho. I will simply make it as an app, and it would wonky work if the app is running.

Because I think we don’t have much resources doing it in the firmware itself.

htotoo, 2025-05-07:
> I’m not sure if the long run stability tho. I will simply make it as an app, and it would keeping work if the app is running.
> 
> Because I think we don’t have much resources doing it in the firmware itself.

we could create a simple script language stuff (exactly the same commands what the serial uses).
so the 'what to do' could be that.
The when could be a simple background thread, that checks the next event time, and sleeps until that. if the timers change, then the thread reloads itself.
would waste around 1kb ram for the thread. for the command processor, i don't know, if we could wire it to serial or i2c command interpreter. (or merge them somehow, ..)

but the overall question is: is it worth it? i think absolutely not. 

so the best way to go is the serial from a raspi or pc

zxkmm, 2025-05-07:
I'm making a external app for this. Maybe someone has a spare PortaPack that they plan to make it open 7*24 doing this 😂