portapack-mayhem/mayhem-firmware #2645: Hard Fault opening gfxEG

AC9DJ, 2025-05-06:
### Describe the bug.

installed n_250506 now I get a hardware fault when opening gfxEQ app.

### Reproduction

Open gfxEQ app.

### Expected behavior

Plays audio.

### Environment/versions

n_250506 on H4M

### Anything else?

Rolled back to ver n_250503 and the app works with no issue.

RocketGod-git, 2025-05-06:
Thanks for reporting. We'll see what happened.

gullradriel, 2025-05-06:
Are you sure you also updated the  external apps and not only the firmware flash ?

gullradriel, 2025-05-06:
It was indeed crashing.

Should be fixed with  https://github.com/portapack-mayhem/mayhem-firmware/pull/2646

AC9DJ, 2025-05-06:
Yes I updated the SD card files.

AC9DJ, 2025-05-06:
Thanks for the update! You guys do some great work, I really appreciate it.

RocketGod-git, 2025-05-06:
@gullradriel  fixed while I got some beauty sleep. thanks! 😎