portapack-mayhem/mayhem-firmware #2653: WFM stereo noise from Portapack board

voxo22, 2025-05-15:
### Describe the problem

guys, it is practically impossible to generate a clean WFM stereo radio signal, either via Portapack (replay) or external software, e.g. GNU radio directly via HackRF, as long as the Portapack board is mounted on the HackRF board. There is a constant hissing/grunting/humming both from the display and from the Portapack itself in the differential stereo channel around the suppressed 38 kHz carrier. I have tried various sample rates (from 200 kHz to 1500 kHz portapack, or 2 MHz - 10 MHz GNU radio), but the interference cannot be eliminated. As soon as I disconnect the Portapack, the signal is only from the Hack RF pretty clean.
Does anyone have experience with this, how to solve it, or is it really necessary to use only the solo HackRF board for these purposes?

### Hardware information

1. 2014, 2020
2. H1, H2

zxkmm, 2025-05-16:
We've tested and most of the interferences are from screen. We prob would implement a turn off the screen functionality for hackrf mode too. And for regular transmitting, you can try if it's getting better when you turn off the screen (by click the moon button)

zxkmm, 2025-05-16:
Note that it's an original designing fault, so we cannot do much fix on hardware side since this device had been sold for years. we can only prevent in software side.

voxo22, 2025-05-16:
yeah, I tried to turn screen off either in Portapack (moon icon), or in hackrf mode (thru debug command), but with no big help, so as you said.. could be HW design problem. Thanks for reply.

closed 2025-05-16 by voxo22

zxkmm, 2025-05-16:
If this is a big thing to you, maybe try take out battery.