portapack-mayhem/mayhem-firmware #2669: Invert rotary encoder

tomventa, 2025-05-26:
### Description of the feature you're suggesting.

In the "Encoder Dial" section, in addition to the multiplier and sensitivity, a flag to invert the behavior of the rotary encoder. This helps especially in case of replacement with a non-original Rotary Encoder EC11

zxkmm, 2025-05-27:
This already implemented for around a year. Please update to latest firmware.

tomventa, 2025-05-27:
Hi @zxkmm , thank you for your reply. I'm currently using version 2.1.0, but I don't see any option to invert the rotary encoder. Could you please provide more details about where to find this setting?

zxkmm, 2025-05-27:
Settings - Encoder Dial - Direction

tomventa, 2025-05-27:
@zxkmm I might have a problem, I don't see this option at all, here are two screen pictures (the menu one only for the version info). Any clue about what's wrong with my mayhem firmware?

[image] 

[image]

zxkmm, 2025-05-27:
ohhh sorry maybe that feature isn't in stable release yet. Last stable release is from 2024.

If you want this feature now, you can update to latest nightly, otherwise you can wait our stable release.

tomventa, 2025-05-27:
@zxkmm I see you implemented this feature with the January 9th nightly, I switched to the latest nightly and it works, thanks!

zxkmm, 2025-05-27:
Ohh sorry I don't remember who implemented this.
But glad that it worked. You are welcome