portapack-mayhem/mayhem-firmware #2704: H2 cannot disable splash screen on boot.

jeffb110, 2025-06-19:
### Describe the bug.

Portapack H2 splash screen always displays at power on. When turned off in setting it turns on again after reboot. Running current nightly release n_250617.

### Reproduction

Power on. Splash screen display. Disable splash screen in settings. Save new setting. Power off. Power on again. Splash screen displays and is reenabled in settings.

### Expected behavior

Splash screen disabled.

tomventa, 2025-06-19:
@jeffb110  
Go to settings -> User Interface -> Uncheck the "Show splash"
If you RTC battery is dead: Return to settings -> User Interface -> P. Memory Mgmt -> Check "Use SD card for P.Mem", click on "Save P.Mem to SD card"
If your RTC battery is ok: Return to settings -> User Interface -> P. Memory Mgmt -> Uncheck "Use SD card for P.Mem"

@gullradriel I have a working RTC battery, but to test it I enabled the "Use SD card for P.Mem", and I don't know if that's normal or not, but settings changes are not saved in real time to the sd card unless you manually go to "P. Memory Management" -> "Save P.Mem to SD card". If it hasn't been implemented yet, it might be a good idea to integrate an automatic saving of settings if the "Use SD card for P.Mem" option is enabled.

gullradriel, 2025-06-19:
> gullradriel I have a working RTC battery, but to test it I enabled the "Use SD card for P.Mem", and I don't know if that's normal or not, but settings changes are not saved in real time to the sd [...]

They should be if the user is getting out of apps using the back button.

You can check in the code, maybe we forgot something.

tomventa, 2025-06-19:
@gullradriel 
> They should be if the user is getting out of apps using the back button.

you're right, if I exit the app before pressing reset it saves my settings, never mind

graealex, 2025-06-22:
Imagine wasting a developer's time with this bogus bug report.

eried, 2025-06-24:
Ok so it is solved 😅

closed 2025-06-24 by eried