portapack-mayhem/mayhem-firmware #2706: Debug > SD Card > Test: M0 Guru Meditation crash

0XDE57, 2025-06-20:
### Describe the bug.

SD card test crashes. But not every time.

Usually I get: 
```
W ms    Fail: Compare
W MB/s 
R ms
R MB/s
```
Looks like the test does not continue after the W ms fail.

But sometimes I get:

```
M0 Guru Meditation

Hint: Hard Fault

r0:  200013C8
r1:  7E20D17E
r2:  00000000
r3:  000726C5
r12: 000000E0
lr:  0007484D
pc:  000747D8
```

Micro SD is ExFat. Card is empty.

### Reproduction

1. Goto Debug > SD Card 
2. Test
3. Repeat until crash

### Expected behavior

1. Test to return some values for W and R

### Environment/versions

v2.1.0

0XDE57, 2025-06-22:
I did try nightly in my second comment: n_250620

Different crash message from the v2.1.0

zxkmm, 2025-06-27:
The crash message is always different among different versions, since the err msg is basically memory addresses.

We'll add protector in the code later.

closed 2026-03-24 by htotoo