portapack-mayhem/mayhem-firmware #2711: DC Performance: improve software between 0-1 MHz

graealex, 2025-06-22:
### Description of the feature you're suggesting.

HackRF is generally capable of receiving signals from DC to 1 MHz, basically the range that is NOT advertised as being available. Performance in that range is poor.

However Mayhem has plenty of glitches in that range that prevent anyone from even trying to receive anything in that area at all. Imho the best course of action would be to not have anything below 1 MHz be treated differently, and instead rely on the user understanding the implications of trying to receive a signal with such low of a frequency.

gullradriel, 2025-06-22:
The HackRF One has an official supported frequency range of 1 MHz to 6 GHz. This means that:
- While it might technically be able to go slightly below 1 MHz, performance becomes very poor and unreliable.
- The analog front-end (the part of the radio that handles actual radio signals) just isn’t designed to work well below 1 MHz.
- Internal components like the RF mixers and filters stop functioning correctly at those ultra-low frequencies.

While Mayhem is showing up or allowing you to tune to below 1 MHz, it's not reliable and may need an up converter to work correctly.

Also:
- Official range: 1 MHz to 6 GHz (as per Great Scott Gadgets, the creators of HackRF).
- Practically usable: Starts to become reliable around 5–10 MHz and up. Performance below ~5 MHz can be very poor or unstable.

Anyway anything regarding hackRF hardware is related to hackRF not Mayhem. See https://github.com/greatscottgadgets/hackrf

gullradriel, 2025-06-22:
> > While Mayhem is showing up or allowing you to tune to below 1 MHz
> 
> Well, that is exactly the problem, it doesn't really allow it, or only when it feels like it. It's not about the HW not being designed for it, just that Mayhem firmware glitches the tuning. I fully agree that the performance is not going to be good.

What do you mean Mayhem glitch it ?
Are you getting better results with hackrf only ? Hackrf Mode?
There are some interferences from the display and internal hardware when using a portapack.
We can not do much firmware side.