portapack-mayhem/mayhem-firmware #2714: ADSB not working in n_250622

plomek, 2025-06-24:
### Describe the bug.

I have noticed my HackRF PortaPack H4M is not picking up any of my local ADS-B signals after updating to the latest firmware n_250622, so I downgraded it to 2.1.0 and ADS-B works great. I suspect this is due to this change:

https://github.com/portapack-mayhem/mayhem-firmware/commit/e26f77ee77c62f67d1e1adf1e0f304b9f081938a  - @tomventa: ADSB database update (tools, db) (https://github.com/portapack-mayhem/mayhem-firmware/pull/2701)

### Reproduction

Using ADS-B RX app in n_250622

### Expected behavior

for ADS-B to work

plomek, 2025-06-25:
After reinstalling the latest nightly version (n_250622), everything seems to be working now. I guess it was just a fluke.

tomventa, 2025-06-25:
@plomek The PR you mentioned was about the database, did you also update the database on the SD card or just the firmware?

plomek, 2025-06-25:
> [@plomek](https://github.com/plomek) The PR you mentioned was about the database, did you also update the database on the SD card or just the firmware?

I just updated the firmware using the Mayhem Hub. I'm not sure if that updates the database on the SD card, but I can't reproduce the issue now.

Brumi-2021, 2025-06-25:
Hi @plomek ,   I have also observed same problem that you pointed , with that nightly n_250622,  (but  n_250617 was ok) .
 But the problem is that,  this problem  it is not happening always 100% . 
It depends on previous launched Applications , and the way you played with them . (sometimes happens , just chaning  step , or tuning up / down , ...)
 So that fail happens  randomly in that nightly  , and therefore it is difficult to catch that wrong behaviour . 
I reset several  .ini settings files,  but so far , no relateion with it. 
Pls check again , in my case ,I  already learnt about how to produce that problem and how to confirm that the problem is gone.. 

But based on my investigations,  tomorrows nightly -that I compiled locally -  will be ok again . 
 (But we need to understand the root cause ). But I will keep checking the new nightlies .