portapack-mayhem/mayhem-firmware #2722: Reverse direction switch for encoder knob

mautas, 2025-06-28:
### Description of the feature you're suggesting.

Some portapacks come with encoders that work OK but in reverse direction, it would be nice to just include a switch  in the encoder configuration section in order to be able to use them.

Phaneron1991, 2025-06-29:
I'm pretty sure what you're looking for already exists.

Go to

Settings

Encoder dial

Direction: normal or reverse

closed 2025-06-29 by zxkmm

mautas, 2025-06-29:
Great! Thank you.

Direction is now fixed. I'm having a hard time getting my configuration to move just once per encoder click. Any advice on that?

tomventa, 2025-06-29:
@mautas I had the same problem, i fixed it by setting sensitivity to low and rate multiplier to 1