portapack-mayhem/mayhem-firmware #2731: APRS GPS tx

f1ghy, 2025-07-02:
### Description of the feature you're suggesting.

Hi, APRS was made to send GPS coordonate. Now, there is ESP32 GPS module. So it could be nice for APRS tx app to be able to send it. I  do not remember exactely how a packet is made, but APRS rx know that as it show a position on the map ;o) So it could be a nice emprouvment... 😉

WXLN, 2025-07-05:
> ### 您所建议的功能的描述。
> 你好，APRS 本来是用来发送 GPS 坐标的。现在有了 ESP32 GPS 模块。所以如果 APRS tx 应用也能发送 GPS 坐标就好了。我不记得具体是怎么生成的数据包了，但 APRS rx 知道，因为它会在地图上显示位置 ;o) 所以这可能是个不错的改进……😉
> 
> ### 还要别的吗？
> _没有回应_

The GPS transmission function uses signal playback function, how can it be transmitted using GPS module?

htotoo, 2025-07-05:
> ### Description of the feature you're suggesting.
> Hi, APRS was made to send GPS coordonate. Now, there is ESP32 GPS module. So it could be nice for APRS tx app to be able to send it. I do not [...]

Could you describe the ui, how you would like it to work?

Like if we integrate the esp's coordinates to the coordinate picker on the map, that would work, but that is a one time set. 
But I can't see, how we could implement a continuous tx of the updated coordinates (like each 30 sec, ..)