portapack-mayhem/mayhem-firmware #2733: Base Station Sonar

Rali0s, 2025-07-05:
### Description of the feature you're suggesting.

https://github.com/StackMaster0ps/mayhem-firmware/tree/Recon-App-Concept/firmware/application/apps/RF%20Sonar%20Mapping%20App

We used the base firmware with GPT to save on Development time for the initial framework of this concept. The GUI does not exist fully yet, however, our README and framework idea and code are above.

Stationary Sonor for Home - Detects Trackers - Built In Failsafe for Blackbox of a house.

-Mike: aka Rali0s w/ StackMaster

gullradriel, 2025-07-05:
I'm waiting for your Pull Request.
It will be funny.

tomventa, 2025-07-05:
@Rali0s You didn't use chatgpt to save time, you used chatgpt to make the readers and reviewers waste time.
I appreciate the attempt, but I really don't appreciate how people can ask chatgpt, get crazy output, not even test, not bother integrating at all, and just make a pointless pull request with useless, broken code.
Even the ascii art in the readme is not well done. "rotate the antenna" what.... you assume that the user has only one antenna and that it is directional? and how do you think you can recognize the orientation without a gyroscope/magnetometer?