portapack-mayhem/mayhem-firmware #2737: ADSB Rx map missing speed parameter

voxo22, 2025-07-12:
### Describe the bug.

I have not tested all nightly versions, but in n250603 (v2.2.0 incl.) this parameter is not shown on the screen. n250328 was ok.
Beside, could you please implement some kind of unit conversion as I can display speed and altitude in metrics?
ADSB Rx is perfect tool for showing realtime flight location and parameters anywhere onboard (tested multiple times and it works like a charm with even small antenna).

### Reproduction

1. just open app and display map on known flight.

### Expected behavior

1. should show speed in upper right corner as before.

closed 2025-07-12 by gullradriel

horrordash, 2025-07-12:
I intended to come up with an acceptable solution and forgot :(

the thing is that now we have IAS, which is also measured in knots, but the numerical value is significantly less than the true speed.
we need to somehow mark on the map that this is IAS. or approximate it to the true speed and also mark this.

Moreover, as data comes in, IAS may change to TAS, but GeoMapView does not allow changing speed units dynamically

htotoo, 2025-07-12:
So we may need to upgrade the map, or convert the values for the map.
Which is the better?

htotoo, 2025-11-23:
yeah, you are right, we should just put it back, and if someone will decide what is the best method, then fix it.