portapack-mayhem/mayhem-firmware #2748: Jammer App Range problems

Castane746, 2025-07-30:
### Describe the bug.

Range issues within the Jammer App since update.  Previously, the app could Jam on all 3 ranges, set within each Range Tab.  Although the ranges can be set with the values the actual transmitting signal (in my case tested with 'Noise') only send Range 1 even when Range 1 is disabled. Range 2 and Range 3 do not transmit their specified ranges.  Also, on the screen to the left side of 'Type'  the width of the frequency range only gives the value of Range 1's width and not the sum total of Ranges 1 to 3 like it did in the previous release v2.1.

### Reproduction

1. A seperate RF spectrum analyser would be required, set to view the ranges to test, to see the bug correctly.
2. Launch the Jammer App by first selecting the Transmit button from the main screen, then select Jammer.
3. I conducted the test within the 2.4 GHz range so I will use the values I used as the example also making sure the values don't interfere with live wifi signals.
4. Select the check box to first enable the range under Range 1 tab.
5. Enter the value 2407 in the Centre box and 5 as the width but leave all other values as default for this test.
6. Select Range 2 and enable the range.
7. Enter 2470 as the centre frequency and 5 as the width.
8. Select Range 3 and enable.
9. Enter 2490 as the centre and 5 as the width.
10. Click the Start button on the HackRf jammer app and use the seperate RF analyser device to monitor the results.

### Expected behavior

1. On the HackRF, the counter at the far left hand side of 'Type' should display a fast incrementing number followed by /15 as the sum total of all 3 Range widths. It only displays /5 though.
2. On the seperate RF analyser device we should see 3 peaks of 5MHz width at each of the stated frequency ranges indicating that the jammer is transmitting at those ranges. Only range 1 is transmitting.

### Environment/versions

This bug is present in Mayhem v2.2 but was okay in v2.1

Danucosukosuko, 2025-08-26:
I can duplicate this issue?

Castane746, 2025-08-26:
> Can I duplicate this issue?

See above under "Reproduction" and "Expected behavior".

zxkmm, 2025-08-27:
should we consider cancel the range tabs? because we have hopper app if you wanna jam long ranges of frequencies.

the range things won't make it hop faster, it was limited to the capabilities of the MCU's calculating speed and how fast PLL can be locked.

the tabs just added frees into the queue anyways, it won't be faster. that separated to tabs.

I mean, can anyone name an actual usage of the 3 range tabs? cuz it looks like mostly useless.

gullradriel, 2025-08-27:
I third.

Castane746, 2025-08-29:
I use it to jam a few separate ranges so it is useful for me and saves time trying to create files for the Hopper app to do this. I would have a go at trying to code it myself but it would take me a long time as I haven't done any coding for decades. But, if no one else uses it there is no point spending any effort on it yourselves. I'll leave the case open for now to see if anyone else has any comments?

htotoo, 2025-08-29:
I'll take a look on it

Castane746, 2025-08-29:
Thanks. Hope it's not too much trouble.

htotoo, 2025-08-30:
Was a tricky bug. After megring this, you will find the fix in the next release.

closed 2025-08-30 by htotoo