portapack-mayhem/mayhem-firmware #2760: Sound board not fixed

Phaneron1991, 2025-08-16:
### Describe the bug.

#2759 Says fixed, but I just updated to latest nightly (n_250816) and I have the same issue as #2758 "no img"

### Reproduction

1. select transmit

2. Select soundboard

Result:

M0 Guru Meditation

Hint: NoImg

### Expected behavior

select soundboard

open soundboard

### Environment/versions

n_250816

### Anything else?

I checked the SD "Apps" folder and it shows the soundboard.ppma and the soundboard is showing up in the App manager on the SD "gray with the SD card icon on the side"

gullradriel, 2025-08-16:
Merge was today, so not yet in existing nightly.
I tested myself with a custom up to date build and it was working.

gullradriel, 2025-08-16:
Wait tomorrow or build a custom next firmware yourself.

gullradriel, 2025-08-16:
If you can't, get into our discord, there is a test ppfw in test-drive.
Flash it using hackrf.app website, select custom and select the file you downloaded from test-drive .

closed 2025-08-16 by gullradriel