portapack-mayhem/mayhem-firmware #2763: new world map

ArminXXXXX, 2025-08-19:
### Description of the feature you're suggesting.

hey friends, when comes a new world map inside the updates? the installation of a better one is horrible. And i am mmissing a wifi rx and tx app for mac adresses. saludos from canary islands 👍

htotoo, 2025-08-20:
World map won't contained inside the update files, since it would take forever to upload and extract.
You need to copy the files yourself.
If you asked when someone generates a new one? Not so much change, so the "old" seems ok.

Wifi rx, wifi tx is not going to happen, unless someone who wants it really, implements it. But i doubt it is possible with the current hw.
Mac addr is for BLE

htotoo, 2025-08-24:
https://github.com/portapack-mayhem/mayhem-firmware/wiki/World-map-generation

closed 2025-08-26 by eried