portapack-mayhem/mayhem-firmware #2773: Jammer Function not working in  v2.2.0

sivaelid, 2025-08-28:
### Describe the bug.

I tried to use the jammer function in the 350Mhz- 440MHz  band using CW  or other waveforms.  Monitoring on a Spectrum analyser the is no output signal. In trying other transmit option  thes work and out put RF signal at Arround -10 dbm. In all cases.

If I load  the previous stable version v2.1.0 then carry out the same activities then the jamming function outputs  RF  as expected.

### Reproduction

 1. Turn on and select the jammer app in the Transmit tab.
 2. enable range 1. I have tried other ranges but did not work either.
 3. Configure the  Frequencies start stop  frequencies and scan rate and other options or  centre frequency and the width.
 4. Select  the rf Gain at say 45  and A on
 5. Press start  and look at output of the sma conector via a protection attenuator on spectrum  analyser .
 6. Should see  spectrum changes.
 7. if you revered to previous stable version of firmware v2.1.0 them the  function works as expected.

### Expected behavior

1. That following about you should see on the spectrum analyser  the generated signal  steping accross the display
2.  signal level varable  and multiple rf siganls depending the waveform selected

htotoo, 2025-08-28:
Could you please test it with the latest nightly?

sivaelid, 2025-08-29:
Yes will do later in the day Sent from my iPhone