portapack-mayhem/mayhem-firmware #2775: AIS Boats - Map view showing wrong speed

gquma, 2025-08-30:
### Describe the bug.

The speed of the vessel is incorrectly shown in the map view. The speed shown in map view does not show the decimal point. A speed of 5.3 knots (shown in the vessel details), is shown as 53 knots in the map view.
See the following screenshots showing vessel details and map view of the same vessel.

[image]

[image]

### Reproduction

1. Open AIS Boats.
2. What until a moving vessel shows on the screen.
3. Select the vessel, and you will see the details of the vessel. Note the vessel speed.
4. Click on "See on map"
5. The speed shown on the Map View screen will be the actual speed X10 as there is no decimal point shown.

### Expected behavior

The Map View screen should show the correct speed with the decimal point

### Environment/versions

n_250828 and previous versions 

### Anything else?

Thank you for all the hard work and development of this project. It just keeps getting better and better!

htotoo, 2025-08-30:
i hope it is fixed.
please try it with the next nightly release, and give feedback. i just assumed the fix, based on other values in the code, but can't test it.

gquma, 2025-08-30:
Thanks very much for the quick response. I will test the next nightly and let you know if it works. 

This was actually the first bug report I have ever made, glad I got it right!

gquma, 2025-09-03:
Hi, the swell has been over 5m around Cape Town, so not many vessels have been able to go to sea. I was able to capture on vessel on Monday and from these screenshots

[image]

[image]

**I see from these screenshots that the speed shown in map view is only the integer part of the number. The value after the decimal point is not shown.**

I will make another post later today to explain how I was able to display all the vessels received on the map, and not just the selected vessel.

I hope this helps for now?